joeycastillo/The-Open-Book #1: Firmware to be used?

Jorgininho, 2019-09-09:
Hello, i love your project but was a bit disapointed when i wanted some code to use.
I wanted to use an epaper to read ebook but can't find good tutorial to explain how to, seems only ebook company manage to display pdf or epub on an epaper.
Maybe i'm wrong, and that you know some already existing project which use custom epaper hardware to display ebook? That you might use for your project?
Or maybe i'm right and you are currently developping one your project? 
In any case i would be glad to have your answer anf maybe your tips

MTRNord, 2019-12-01:
I am wondering if there is a way to use this: https://github.com/koreader/koreader I never tried it myself but the screenshots look quite intresting

tnyeanderson, 2019-12-23:
Please look into this as well: https://github.com/bq/cervantes

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.