joeycastillo/The-Open-Book #2: Questions

mistergibson, 2019-09-14:
Dear Mr. Castillo,

1.  Will you use the e-Ink electronic paper at 1200+dpi?

2.  Have you considered using mRuby (embedded Ruby) ( Found Here: https://github.com/mruby/mruby ) in place of the Python selection?

3.  What are the best batteries to use? : I get about 2 weeks out of my Kindle-Paperwhite.

I just also wanted to say Thank You for doing this. I have long wanted an ebook reader unbound to any one corporation. I would spend $150 - $200 to get one that is well designed. I don't need color, and I don't want to watch movies >>> what I want is a high-dpi book-like experience in an ebook reader independent of any one corporation. If you do an at-cost business around this idea please also consider accessorizing with a small USB-based Solar charger for it. You've inspired me just by doing this project. Good Luck to you.

mistergibson

Dragoon91786, 2019-09-14:
Furthermore, if the project decides to use E-Ink/E-Paper, will it use the full color format? httpss://goodereader.com/blog/electronic-readers/e-reader-companies-will-adopt-color-e-paper-in-2018

Dragoon91786, 2019-09-14:
@mistergibson, have you heard about the newer Li batteries that don't wear down over time to any degree as the older ones?

There's also new rechargable alkaline batteries.

I saw in 2012 tech from Northwestern that was using crystals as well. Another possibility would be newer https://en.m.wikipedia.org/wiki/Solid-state_battery as an option if the device uses a universal standard battery plug. One factor with these devices is the lack of universal battery comparability. It would be nice to have a device that could use various sizes and batteries available if necessary. Not necessarily ideal in some minds, but if it could act as a more sophisticated version of the AK-47 in terms of ammo (but from the standpoint of a battery), it would definitely make the devices more universal. Much like how one can run a D battery with adapters using C, B, A, AA, AAA, etc.

joeycastillo, 2019-10-31:
> Will you use the e-Ink electronic paper at 1200+dpi?

At this time I am targeting lower-resolution, inexpensive displays that are widely available. I do not know of a source for 1200dpi screens. 

> Have you considered using mRuby (embedded Ruby) ( Found Here: https://github.com/mruby/mruby ) in place of the Python selection?

I have not, currently my experience is in Arduino and CircuitPython, but you're more than welcome to hack on mRuby if you're stoked about it! 

> What are the best batteries to use? : I get about 2 weeks out of my Kindle-Paperwhite.

I'm currently using [this battery from Adafruit](https://www.adafruit.com/product/3898) because it fits nicely between the Feather headers. Eventually when an enclosure is built it might make sense to have room for a larger battery, but I have no sense at this time what battery life will be like. 

> if the project decides to use E-Ink/E-Paper, will it use the full color format? 

I doubt it as I have not seen a source for full-color eInk panels. I have done tests with some tri-color panels, but their refresh rate is far too slow to allow for a good reading experience.

closed 2019-10-31 by joeycastillo