joeycastillo/The-Open-Book #5: Will the screen size be an option?

bmix, 2019-10-10:
Hi,

congratulations and thanks for a very nice effort! 

May I suggest to use a bigger screen size? The biggest problem with current eBook readers, from the *usability point of view*, are the bad support for PDF, which still is the format, most technical and scientific papers are being published in. PDF doesn't float, so a screen size >8" would be needed for comfortable consumption.

I am desperately seeking for a reader, that supports ePub, PDF and (X)HTML (even CHM and CBA would be nice), so much, that I even would bother to solder my own.

Maybe supporting more than one screen size could be an option?

anjannath, 2019-10-11:
> Maybe supporting more than one screen size could be an option?

Since, this is an open source project, i guess most of the things would be customizable.

One more think about the screen, it should use the eInk displays that comes in the kindles.

the-loan-wolf, 2019-10-14:
Kindle paperwhite screen is really good it has 300ppi so font look sharp and give good reading experience

joeycastillo, 2019-10-31:
Hi and sorry for the delay in responding, I'm just now catching up on old issues. I've gone over some of the screen size considerations [in this thread](https://github.com/joeycastillo/The-Open-Book/issues/10#issuecomment-548576209); the short answer is that there's a 7.5 inch display that's expensive but probably compatible, and that larger screens are both expensive and not immediately compatible. We'll see what the future holds.

closed 2019-10-31 by joeycastillo