joeycastillo/The-Open-Book #8: Contribution Guide

hazkaz, 2019-10-15:
Hi @joeycastillo , this is a great effort and idea. It would be nice if you could put up a contribution guide for people like me wanting to work on this project. Thanks.

ech0r, 2019-10-15:
I would like to contribute as well. I came across this project on Hacker News.

titimoby, 2019-10-15:
How do you think we can help you?

To have more code contribution quickly, do you think a simpler device like only a feather M4 with a e-ink wing directly from adafruit could be a first base to help you?

hazkaz, 2019-10-17:
How do we find out what other boards will be compatible with the software? Since the hardware is likely to change in the future, how do we make the code more compatible with other hardware?

titimoby, 2019-10-17:
Don't be sorry, as I can see, there is quite a lot of people interested and you have many questions ;)
I'm sure you need some time to have a proper hardware you are happy with and I will wait.

But in the meantime, I was wondering if existing pieces from Adafruit can be put together to start messing with software: I mean something usable even if it is on a breadboard with plenty of wires.

joeycastillo, 2019-10-17:
Actually yes! Mostly. There are a couple of parts you'd need from elsewhere (namely the [screen](https://zh-tw.buyepaper.com/42inch-e-ink-panel-spi-interface-buy-e-paper-display-p0085.html) and a spare [SPI Flash chip](https://www.digikey.com/product-detail/en/gigadevice-semiconductor-hk-limited/GD25Q16CTIGR/1970-1010-1-ND/9484760) for Babel stuff), but beyond that you can build the equivalent of the wing with: 

* 1x [eInk friend](https://www.adafruit.com/product/4224)
* 1x [microSD breakout](https://www.adafruit.com/product/254)
* 1x [MCP23008](https://www.adafruit.com/product/593)
* 1x [mono headphone jack](https://www.adafruit.com/product/4031)
* 1x [SOIC-8 breakout](https://www.adafruit.com/product/1212) (for the Flash chip)
* plus some [buttons](https://www.adafruit.com/product/1489) and a [switch](https://www.adafruit.com/product/805)

But if you're comfortable with the soldering, honestly I'd recommend just building the wing; it's fun to hold and makes hacking on it a joy. If there's enough interest I might make a small run of like 10 of Rev C just to get them out into the world; aside from the 24-pin header, which I could probably reflow here, all the rest of the soldering is pretty doable by hand. It's all SOIC chips and 0805 surface mount, nothing too intimidating.

titimoby, 2019-10-18:
I have no issue with soldering even if I only know hand soldering until now.
This project could be a good first for smd soldering.
Let me know if you decide to launch a batch of Rev C, I'll be glad to buy one and experiment.

liamdawson, 2020-01-02:
I have zero electronics experience, are there any simulators I could try things out in? I was once investigating the possibility of doing firmware for a different SoC (RTL8710AF) in Rust, would be fascinated to see if I could apply that idea in this scenario.

AndersCulverD5, 2020-02-02:
@joeycastillo I came across this project recently and I'm also interested in helping in some way. I'm a junior embedded software engineer and I've been looking for cool projects to contribute to. Is the list of parts needed to do development the same as your previous reply from October? Let me know if there is something that I could try to hack on!

beelzebielsk, 2020-03-08:
I am also interested in this project, and in contributing to it. I recently bought a kindle, and while I like the experience, I want something open that I can hack on. I realized my reading device is a very personal thing, as my reading experience is.

I have no electronics experience, but I am a junior software engineer.

braininahat, 2020-06-20:
I too would like to contribute to this project. I have experience in all sorts of things from embedded to computer vision and can do software architecture design stuff too. Is there a mailing list or something where you discuss the roadmap?

joeycastillo, 2020-06-20:
Hey all and I'm so sorry for dropping the ball on this thread. It's been a strange year. 

The situation as it stands: I have PCBs of the E-Book FeatherWing [available on Tindie](https://www.tindie.com/products/joeycastillo/the-e-book-wing-pcb-bare-pcb/); this is a bare PCB that, when assembled, is designed to work with a Feather M4 Express to enable most of the functionality of the Open Book. I chose to make this design available first since it's a simpler build. The BOM is [available on KitSpace](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/), and with their browser extension it's one click to order almost all of the parts on DigiKey. 

I also have PCBs of the Open Book that I plan to make available later this week on Tindie; I just want to finish up some documentation and the software necessary for flashing the bootloader and testing the device. The boards will show up [on this page](https://www.tindie.com/stores/joeycastillo/) when available. And the same Open Book library in this repository should work seamlessly on both the E-Book Wing and the Open Book.

While it's not quite a contribution guide, I'm working on some [rather extensive documentation on the Open Book](https://github.com/joeycastillo/The-Open-Book/wiki/Open-Book-Documentation); as of now it's 12,000 words and covers both tips for building the device, and the documentation you would need to start developing for the device. I'll also be making a final BOM with one-click ordering available on Kitspace in the coming days. 

I've been focused on getting the PCB's available and writing basic software drivers for things like the e-paper display and the [language support](https://github.com/joeycastillo/babel), but I have not had as much time to dedicate to the e-book reading software. There is [a demonstration sketch called MVBook in this repository](https://github.com/joeycastillo/The-Open-Book/tree/master/Examples/Open_Book_MVBook) that shows a sort of minimum viable prototype for what I hope the software will look like ([video here](https://twitter.com/josecastillo/status/1220433844035297280)). I have some ideas about how I might want to architect the software, and opinions about what texts should look like (plain UTF-8 text with simple YAML or TOML front matter), but no formal roadmap for this. 

I am hoping to make moves on this effort in the month of July, including documentation around the software effort and a more formal guide for contributors; I had hoped to be far further along by now, but I'm afraid I've been more scattered this year than I would have liked. In the meantime I welcome any suggestions or contributions in this thread, and will respond here as soon as I have the Open Book PCB's available for folks who want to build their own.

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo