joeycastillo/The-Open-Book #9: Migrate to KiCad for PCB design?

foozmeat, 2019-10-30:
Hi, great project idea and I'm excited to contribute. However, I'm not excited to use Eagle. KiCad is mature enough to handle this project and is FOSS. Would you consider moving to it?

joeycastillo, 2019-10-31:
I would be thrilled to move to KiCad! I know that it being in Eagle is a barrier to people who want to contribute, but don't want to or can't pay for a monthly subscription. For now though, Eagle is the tool that I know and that I can use to iterate quickly. Once I get the design stable, I'd love to learn KiCad and remake the project there, but it might be something that happens later, once I've got a revision of the board finalized.

keen101, 2020-01-26:
I don't use kicad, as I also learned and use Eagle, but i believe kicad has a tool to import and migrate eaglecad designs to some degree. Not sure how well it works,  but might help save some time recreating from scratch. 

https://kicad-pcb.org/blog/2017/10/Testing-Eagle-import-plugins/

LogoiLab, 2020-06-20:
It is sort of a waste to have this be open source if it requires fairly expensive software to contribute to/utilize. KiCad isn't that bad. I use it all the time, definitely usable, you dont get all those crazy macros and such that youd get with something like Altium, but it works well.

jupart, 2020-06-21:
I could help with the migration. I use KiCad at work designing/changing 20-30 PCBs a year.

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo