joeycastillo/The-Open-Book #11: Will you be selling these?

nerdguy1338, 2019-10-31:
I love this concept, but is this a design to hack upon, or plans for a final product that is open hardware?

joeycastillo, 2019-11-01:
That's a great question, and one for which I don't have a great answer just yet. I can assemble prototypes myself, but that doesn't scale well enough to sell them in any quantity. I guess my vision is that people would build these themselves — you can do it with a stencil, tweezers and [a toaster oven](https://learn.adafruit.com/ez-make-oven?view=all) — but I'm also willing to look into manufacturing. It's not 100% in my wheelhouse, but then neither was PCB design a year ago :)

nerdguy1338, 2019-11-01:
That's the thing, most of the board you designed is through-hole, except
the feather itself. I've never done surface mount soldering, and I'm not
really comfortable gambling this much money on a kit of parts.

Why is the feather not also through hole?

joeycastillo, 2019-11-01:
Basically because the e-ink screen is fragile glass, and needs to adhere directly to the the front of the PCB for structural support. Most of the board is surface mount actually, specifically so that the front of the PCB can be a smooth and flat surface for the screen to sit on; all the thru-hole pieces are above and below the screen. 

Totally sympathize though with your concern... I was also very intimidated by surface mount soldering and had to learn some new skills to make this. FWIW except for two tricky parts (the SAMD51 and the 24-pin connector) I chose wide-pitch SOIC chips and 0805 or larger passives, so that it would be as easy as possible to pick and place by hand. It's not as easy as thru-hole, but it's definitely doable, especially if you practice on some simpler surface mount boards to start with.

AsherGomez, 2019-11-02:
Hey, awesome project! I like that it works with a Feather ecosystem. You should submit it on Hackaday.io for this [contest](https://hackaday.io/contest/168107-take-flight-with-feather) that’s running right now, good luck!

nelsonblaha, 2020-04-14:
Someone willing and able to make these to-order should name their price here. I think they would be pleasantly surprised at the demand.

gromain, 2020-04-20:
I believe those could be made by quantities 100 for between 150-200$ piece (complete assembled with casing and shipping).

1094, 2020-05-07:
I understand the not wanting to sell completed readers, but have you considered selling kits?

joeycastillo, 2020-05-08:
It's not the Open Book per se, but I did have a run of the E-Book Featherwing PCB's made, and as of today [they are available on Tindie](https://www.tindie.com/products/joeycastillo/the-e-book-wing-pcb-bare-pcb/). I started with this board since It's simpler to build than the Open Book, and if you pair it with a Feather M4, it works seamlessly with the Open Book Arduino library. Not quite a kit, but the BOM is included as a spreadsheet you can upload to Digikey, and the only other part you'd need is the e-paper display which is [available from Good Display](https://zh-tw.buyepaper.com/42inch-e-ink-panel-spi-interface-buy-e-paper-display-p0085.html).

I imagine I will likely do a similar run with the Open Book PCB at some point, but this is what I have today :)

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo