joeycastillo/The-Open-Book #12: Constructive Comment

AsherGomez, 2019-11-02:
Hey, awesome project! I like that it works with a Feather ecosystem. You should submit it on Hackaday.io for this [**contest**](https://hackaday.io/contest/168107-take-flight-with-feather) that’s running right now, looks like a great opportunity for this to grow *fast*. Good luck!

joeycastillo, 2019-12-03:
Thanks for letting me know about this contest! It took a few weeks but I got the project into shape and submitted it here: https://hackaday.io/project/168761-the-open-book-feather

closed 2019-12-03 by joeycastillo

justinclift, 2019-12-04:
@joeycastillo As a data point, the image on the mailing list page you mention on the Hackaday submission is 404.  eg going to the [mailing list page](http://eepurl.com/gKOpQ9), gives a broken image banner for the top 1/2 of the page.  (Guessing it's a large image :wink:)