joeycastillo/The-Open-Book #16: Donations accepted?

tnyeanderson, 2019-12-23:
I would love to support this project, and money tends to be helpful when it comes to open source. Is there a way to donate to this project to help it along?

Thank you for doing this, Joey. This is a glaring hole in the open source world, and we desperately need open ereaders... Your work is very much appreciated!

joeycastillo, 2019-12-28:
Thanks for the kind words! Since a lot of people have asked, I made [a Patreon page where people can donate](https://www.patreon.com/joeycastillo). Also seems like a good place to post day-to-day notes and work in progress stuff.

justinclift, 2019-12-29:
Good idea. :smile:

closed 2020-01-01 by tnyeanderson

616b2f, 2020-01-01:
@josecastillo maybe you could use this new GitHub feature: 
https://help.github.com/en/github/administering-a-repository/displaying-a-sponsor-button-in-your-repository 

This would be more convenient :)