joeycastillo/The-Open-Book #18: Great Job!!!. What about of think on linux?

wkrea, 2020-01-21:
First of all I want to congratulate you for an excellent idea and purpose of your work.

In another hand, i think that is a good time for suggest and offer some help to you on this project, for improve oriented to the final user to future.

Really, make a board and desing all firmware it's a great job for engineering and intelectual purposes, but in long way it's very difficult for manteniance and flexibility, in way to give for give more functionalities to ereader.

Have you think about in more storage, conectivity and compatibility?. there are many boards (open hardware) than should be great for supply this requierements, which need more people workning on firmware for eink display.

For not go far with my idea. What if think on some board that support linux and give conectivity on wifi or bluetooth? and only give mode suport for eink displays in very size and resolutions?.

This give to the final user, more flexibility for have an ereader with an app of your preferences for read. More flexibility for us for think on the Ux/Ui and other funcionalities for pluralize the ereaders in low cost and good quality..

Let me know what you think about this point of view, if your purpose is oriented to this or is different?

one more time, great job men. You are a master!!!

torpak, 2020-01-22:
Disclaimer: I have no connection to this project, (other than finding it awesome) and have not talked to the author.
But, it seems to me that one of the goals was keeping the cost per unit managable. Using powerful hardware (as in powerful enough to run Linux) would significantly increase the price tag for small batches.
Also there is something to be said for Simplicity. An ebook reader that only does what is neccessary for it's purpose is a much simpler system, than one that carries a whole operating system with it. Simpler systems have less failure modes and are easier to get right (especially if you have limited manpower).
The last aspect ist battery usage: a powerful system needs a powerful battery, again driving up costs and consuming resources.

ngsctt, 2020-01-22:
There's some discussion about Linux in #10, including the possibility of using the [Giant Board][gb] and some of the limitations of doing it that way (including cost), about how to add WiFi.

[gb]: https://www.crowdsupply.com/groboards/giant-board

joeycastillo, 2020-01-25:
Thanks for the kind words! I've actually ordered a Giant Board with the specific intent of exploring Linux options with the Wing. I also think it could be interesting to engineer something that pairs with a Raspberry Pi or compute module. For all the reasons outined in #10 though — more expensive, less DIY friendly parts, and more expensive manufacturing processes — I think designing a full Linux board is probably out of scope for this project. 

Keep in mind that even at scale, many full-featured Linux-based e-readers [are sold at a loss](https://www.pcworld.com/article/243531/report_79_amazon_kindle_costs_84_to_make.html); the companies that sell them make up the difference by stealing your attention with ads and selling DRM-encumbered content. I want to keep the cost low and avoid all that nonsense, which means making some tradeoffs.

Thanks for the suggestion though, and again I encourage anyone with this kind of interest to fork the repository and play with the design; it's all open source :)

grravity, 2020-07-26:
I world really hope for an ability to just rsync my directory which I already have rsynced between my computers, but also just have it rsync to the open book. If this is possible I'd be happy.

GarkGarcia, 2020-09-04:
> I world really hope for an ability to just rsync my directory which I already have rsynced between my computers, but also just have it rsync to the open book. If this is possible I'd be happy.

I believe you can simply mount the microSD (the one used for storing the books) on your machine and rsync the directories from from there.

grravity, 2020-09-08:
> I world really hope for an ability to just rsync my directory which I already have rsynced between my computers, but also just have it rsync to the open book. If this is possible I'd be happy.
>
> I [...]

The issue with this is, it would not be possible wirelessly. I can easily copy over files if it's mounted.
But that would require hooking up everytime I add new books.
Interfacing with Calibre wirelessly could be another option.
I know it's not the most preferable feature to always have wireless on though.

wa2mze, 2020-11-11:
The hw cost difference between using a RPi Zero (with wifi/bt) and the samd51 feather actually might favor the RPi (would have to mount the RPi 40 pin header to the BOTTOM of the board so it can plug in to the main board).  The software development effort would involve different skill sets than the Arduino based project, a good working knowledge of GDB would help!  But once you go down the Linux rabbit hole the project WILL end up being more of a PDA or Smart Phone, then an ereader, and the eink display would become a liability because of its slow update rate.  Still, Linux will buy you 'free' OGG music file support (I don't think there is an arduino lib for that yet), and free support for most epub formats (free meaning it's already mostly written).
BUT, I don't think that the current SAMD51 based open book is the wrong approach.  You will get lower power drain, longer batter life, and you CAN build the whole thing yourself.  The Linux approach will require using a pre-built CPU module (RPi Zero, RPi compute module, Beagle Bone submini, Giant Board, etc).  There is room for both (and I have a Pi zero I don't know what to do with at the moment), but I'd like to build the Open Book as is stands, though I hope some PDF or EPUB file support will be ported in eventually.

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo