joeycastillo/The-Open-Book #20: Sister project? Inkplate

JustGitting, 2020-02-13:
Hi,

It's exciting to see the Free-Software / Open Hardware movement gain momentum with the explicit aim of protecting our privacy. 

I just found your project and see that you have very similar goals as [Inkplate](https://github.com/e-radionicacom/Inkplate-6-hardware), also an open source e-paper project. 

1) Would you be able to make a comparison between your project and Inkplate?

2) Would you consider helping each other as you both are working on essentially the same project? (I'm not affiliated, I just want both of your projects to succeed!)

Best wishes.

joeycastillo, 2020-02-13:
Hesitant to make a point by point comparison (it's a great project and I'm following it with interest!) but the short answer is that they're focusing on connectivity (wifi and bluetooth), and I'm focusing on local peripherals (lots of buttons, audio in/out, Feather/STEMMA compatibility).

They're also using a more expensive, higher-resolution display that has some more complex power requirements, including an external power management IC for the screen. I'm excited to see how their work on that evolves, because while it would likely blow up the BOM cost for this version of the Open Book, I'd love to design a more powerful version at a later date, and I'd be open to any exchange of ideas with other OSH developers working on similar projects. 

Thanks!

davidzovko, 2020-04-06:
Hey @joeycastillo, great to hear that you laid your eyes at Inkplate :) I'm David that's behind the Inkplate project. 

The final schematics and BOM has been just [published](https://github.com/e-radionicacom/Inkplate-6-hardware/tree/master/Schematics%2C%20Gerber%2C%20BOM). Maybe you will find something useful :) I am happy to talk furtherer about our projects, if you are interested, too.

jackbrennan-creator, 2021-05-30:
This simple eBook reader for InkPlate might be of interest. [minimalist-ebook-reader](https://github.com/jackbrennan-creator/minimalist-ebook-reader)
Also a port to the M5Stack Core 2. Just to make one you need a magnifying glass to read! [ebook-reader-for-m5stack-core-2](https://github.com/jackbrennan-creator/ebook-reader-for-m5stack-core-2)

davidzovko, 2021-06-01:
Very cool, thank you for pointing these out!

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo