joeycastillo/The-Open-Book #24: How to configure/compile library for other hardware

michalsok, 2020-03-24:
Hi. I purchased my EPD (4.2") and Adafruit Feather M0 logger (with SD card socket) to build my simple eBook reader. Then, while searching for libraries to simplify my project, I stumbled across your project and was super excited that I don't have to start from nothing.
My setup is simple: EPD and SD on SPI plus MCP23008 for buttons in I2C
I downloaded the folder and modified the OpenBook.h like so:

`#elif defined(ADAFRUIT_FEATHER_M0) // e-book wing on Feather M0
#define OPENBOOK_DISPLAY_BUS (&SPI)
#define OPENBOOK_SRCS (-1)
#define OPENBOOK_ECS (10)
#define OPENBOOK_EDC (9)
#define OPENBOOK_ERST (6)
#define OPENBOOK_EBSY (5)
#define OPENBOOK_SDCS (4)
#define OPENBOOK_BCS (-1)
#define OPENBOOK_BATTERY_MONITOR (A7)
#define OPENBOOK_BUTTON_INTERRUPT (11)
#define OPENBOOK_BUTTON_LATCH (-1)
#define OPENBOOK_BUTTON_DATA (-1)
#define OPENBOOK_BUTTON_CLOCK (-1)
#define OPENBOOK_BUTTON_LOCK (-1)
#define OPENBOOK_BUTTON_ACTIVE (LOW)
#define OPENBOOK_AUDIO_L (-1)
#define OPENBOOK_AUDIO_R (-1)
#define OPENBOOK_MIC_RAW (-1)
#define OPENBOOK_MIC_AMPLIFIED (-1)
`
Disabled the serial RAM as this waveshare 4.2 BW already has one built in
and I also wanted to disable Flash for Babel.
I am assuming that it is the OPENBOOK_BCS (-1) that does it. (Don't have EAGLE so I can't verify)
I am building a simple external PCB with MCP23008 and buttons to handle buttons.
How do I disable Babel support (can it be disabled?)
When compiling I get an error: "BabelTypesetterGFX.h: No such file or directory"
What is causing it?
Could you help me to modify the library setup to work with my simple reader hardware?
Thank you,
Mike.

joeycastillo, 2020-03-25:
Thanks for bringing up this issue, it's something I hadn't gotten to yet. One of my goals has been to make a "lightweight" version of the Babel library that just includes ASCII glyphs in memory, to allow support a consistent method for typesetting even if you don't have the extra chip installed.

For the moment, if you just install the library ([from here](https://github.com/joeycastillo/babel)), it will silence the errors, but I may need to figure out a better way to do this for folks that want to pick and choose specific bits of functionality.

michalsok, 2020-03-26:
Thank you. I will give it a try.

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.