joeycastillo/The-Open-Book #28: Slightly larger, high resolution display?

Drizzt321, 2020-06-23:
So love, LOOOOOVE this. Will definitely be making one of these, although my trust PRS-T3 is managing to stay alive. Personally I prefer the size of a 6" display, rather than 4.2". I found https://zh-tw.buyepaper.com/6-inch-1024758high-resolution-epd-display-p0096.html except it's 34-pin, so not a drop-in compatible. Is there any info that, in your research, you've found that could be useful to someone wanting to tweak the design to support these slightly larger displays?

joeycastillo, 2020-06-24:
A screen with a parallel interface is something on my radar screen, though the two panels I'm more interested in are either [Good Display's 4.3 inch high resolution panel](https://zh-tw.buyepaper.com/43-inch-low-power-consumption-e-paper-p0093.html) or, more exciting to me, using a recycled Kindle screen like the [Inkplate 6](https://www.crowdsupply.com/e-radionica/inkplate-6) does. The Inkplate folks' Github repository has [some code related to driving that screen](https://github.com/e-radionicacom/Inkplate-6-Arduino-library) that I want to play with, once I get software support for this iteration of the Open Book further along. 

Also, note that those panels, to my understanding, do not have an on-board booster, which means you would likely also need to add a PMIC like the [TPS65185](https://www.ti.com/lit/ds/symlink/tps65185.pdf) or [TPS65186](https://www.ti.com/lit/ds/symlink/tps65186.pdf) to generate the higher voltage rails necessary to drive the screen.

As for the 5.83 inch screen, I've tested it with the Open Book (using the [GxEPD2 library](https://github.com/ZinggJM/GxEPD2)), and it "just works"; the same booster circuit and flex connector gets images displaying on the screen. As you say it would involve some dimension tweaks and a larger cutout, but relatively trivial. Having said that, a couple of reasons I don't think it's a great match: 

* at $40 per unit, it's at least twice as expensive as [the 4.2 inch screen](https://zh-tw.buyepaper.com/42inch-e-ink-panel-spi-interface-buy-e-paper-display-p0085.html) in small quantities
* slower screen refresh
* no support for partial refresh or grayscale modes

I think it's rad to be looking at larger and higher-res screens, but be careful to watch out for both price and features in addition to panel size and resolution :)

Drizzt321, 2020-06-24:
No, didn't notice the slower refresh and no partial-refresh. Yea, those are big features for e-books. That plus the cost makes that 4.2" far superior.

Hadn't seen that Inkplate 6, very cool. Nice to see that you should be able to re-use some of the same software. A popular quality screen like for a Kindle definitely is useful to have support for. I've got some other projects, but hoping I might be able to join in this winter after I clear some of them off. And get more Arduino dev experience.

seniorm0ment, 2021-04-14:
Any updates on this?
I'm interested in building an Open Book, that being said the small display is a complete deal breaker for me currently. Of course a larger screen would imply a higher cost, that being said having the option is nice. 
I think it's important to try to compete with retail readers such as Kobo and Kindle.

An 8" 1440 × 1920 300 PPI display like the one found in the [Kobo Forma](https://us.kobobooks.com/collections/ereaders/products/kobo-forma) is **ideally what I'm looking for**.
If not that, then _at least_ a 6" 1072 x 1448 300 PPI display similar to the one found in the  [Kobo Clara HD](https://us.kobobooks.com/collections/ereaders/products/kobo-clara-hd).

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo