joeycastillo/The-Open-Book #30: C2 - missing from Feather BOM

chadvavra, 2020-07-15:
https://github.com/joeycastillo/The-Open-Book/blob/master/eBook%20Feather%20Wing/1-click-bom.csv

C2 is not included in the 1 click BOM

joeycastillo, 2020-07-15:
Oh wow, that is an oversight on my part. I'm shocked I didn't catch it. It's supposed to be a 10µF capacitor identical to C1. 

Fixed in 27c5aa9; if you bought parts and are one short, I'd suggest leaving C1 unpopulated and putting the 10µF capacitor in C2's spot, though it might work either way. So sorry about that, and thank you for raising the issue!

closed 2020-07-15 by joeycastillo

chadvavra, 2020-07-15:
no problem.  I did end up one short, but my screen is at least a week out and Amazon has cheap sample packs of 0805 caps available via Prime.