joeycastillo/The-Open-Book #31: Adafruit Feather STM32F405 Express

chadvavra, 2020-07-16:
https://www.adafruit.com/product/4382

Thoughts on using the STM32 instead of M4 Express?

We tested this in Arduino [STM32duino](https://github.com/stm32duino) with all our FeatherWings and only the RFM69/RFM9x libraries did not work

STM32F405 Cortex M4 with FPU and 1MB Flash, 168MHz speed

- 192KB RAM total - 128 KB RAM for general usage + 64 KB program-only/cache RAM
- 3.3V logic, but almost all pins are 5V compliant!
- USB C power and data - our first USB C Feather!
- LiPo connector and charger
- **SD socket on the bottom, connected to SDIO port**
- **2 MB SPI Flash chip**
- Built in NeoPixel indicator
- I2C, UART, GPIO, ADCs, DACs

joeycastillo, 2020-07-17:
Initial thoughts are that it should work! Early on I tested the wing with M0 Feathers as well as the the ESP32 Feather, and at the very least the screen test sketch worked as expected. I have had an STM32 Feather here for a while, but haven't had a chance to test it. If you wanted to try, you could more or less copy over [the hardware definition for the M4 Feather](https://github.com/joeycastillo/The-Open-Book/blob/6232c1c80b1a5533d58d1272009904f260bea406/src/OpenBook.h#L67) since almost all the pin numbering is the same.

The only issue I foresee is that the BCS pin, which has a solder jumper to allow it to connect to the M4's 'Extra' pin, would need to be connected to a different pin since the STM32 Feather uses it to enter bootloader mode. If you planned to use the SD socket on the Feather instead of the SD socket on the wing, you could run a wire from the BCS pad to SDCS, and then use pin 5 to select the language chip.

If you try this, definitely let me know in this thread! And either way I'll likely give it a go in the coming weeks.

chadvavra, 2020-07-17:
Will do!  Thanks.  I'm curious about the chip speed and if it makes any difference.Sounds like it might be hard to try both out on the same board though.

joeycastillo, 2020-07-18:
Yea, if you connected the two sides of the BCS solder jumper, like it suggests in the getting started guide for the M4, it'll probably interfere with things since that pin is pulled down on the Feather; the Flash chip will always be selected in addition to whatever device on the bus you're trying to talk to. You can un-bridge the jumper with some solder wick and run a wire, but then it's a solder job every time you want to switch Feather boards. Not ideal.

nerdguy1338, 2020-07-18:
Make that jumper a pushbutton switch instead?

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo