joeycastillo/The-Open-Book #32: FPC connector insertion is pretty hard

justinBR, 2020-07-28:
Hi,
Great project! I've been having a really hard time getting the flex cable from the pcb into the FPC connector. I used the fpc connector from the BOM, not the one the that came with the display (I didn't have it when I assembled the board). I can only get it to insert if I push it in from an angle > 30 degrees. because the cable is so short, it does not seem possible to get the cable to this angle when it is threaded through the hole in the PCB. Is there some trick to getting it to insert, or do I need to remove the connector and replace it?

justinBR, 2020-07-28:
it is doable, just hard. sorry for raising an issue for such a goofy thing.

closed 2020-07-28 by justinBR

joeycastillo, 2020-07-28:
No worries, not goofy at all! It's a bit fiddly, for sure, and I should have anticipated that it would be a friction point. First, make sure that the gray tabs are open. Then as far as fitting the connector, I've found it helps to kind of gently bend the cable in between the connector area and the ground pour before you thread it through the hole; that gives you some leverage to work it in, like so: 

[image]

A little bend like this shouldn't damage the flex cable, especially since once it's in, you aren't going to be flexing and unflexing it repeatedly. And then make sure to close the gray tabs once it's in place.

For anyone else using the Good Display connector, the method is the same except that instead of gray tabs it has a black lever on the opposite side from the cable; that lever needs to be flipped up for insertion, and pushed down once the cable is in place.