joeycastillo/The-Open-Book #34: Testing out an interactive assembly guide on kitspace

kasbah, 2020-08-29:
Hey, we are testing out the interactive assembly guide for Eagle projects on https://github.com/kitspace/kitspace/pull/300 (by converting to KiCad and using the Interactive HTML BOM plugin for that)

Here's what it will look like:

- [Open-Book](http://eagle-ibom.preview.kitspace.org/interactive_bom/?github.com/joeycastillo/The-Open-Book/Open-Book)
- [eBook-Wing](http://eagle-ibom.preview.kitspace.org/boards/github.com/joeycastillo/The-Open-Book/eBook-Wing/)

This look good to you? Do you see any issues?

joeycastillo, 2020-09-04:
It looks great! One thing is that there are some board features (fiducials, logos) that don't have a part associated with them, but still appear in the list. If there's a way for me to tag them to be excluded, I'd love to do that to make it clearer to folks.

kasbah, 2020-09-05:
Great, we are just trying to improve the silkscreen text placement before we put this up on the main site. 

Regarding excluding components, I am not sure. In KiCad you can mark components as "virtual" to exclude them manufacturing outputs. Is there an equivalent for Eagle?

kasbah, 2020-10-15:
It's live now!

https://kitspace.org/interactive_bom/?github.com/joeycastillo/The-Open-Book/eBook-Wing

and

https://kitspace.org/interactive_bom/?github.com/joeycastillo/The-Open-Book/Open-Book

closed 2020-10-15 by kasbah