joeycastillo/The-Open-Book #37: New gerber/drill files broke kitspace page

kasbah, 2020-09-14:
Hey, moving the Gerbers and drill files into separate folders isn't supported by Kitspace yet. Your pages currently won't update any more. 

Options are:
- Move the drill and gerber files into one folder and update the kitspace.yaml
- Wait till we add support for it: https://github.com/kitspace/kitspace/issues/12

joeycastillo, 2020-09-14:
Weird, I updated the files but I thought the structure was the same (it's just the directory structure that Eagle outputs). If I just move everything to the root of the dated folder it should work though?

kasbah, 2020-09-14:
You are right, it is the same as before and it's building fine now. Not sure what happened

What I said about the separate folders is true though and currently the gerber download zips are missing the .xln drill files.

joeycastillo, 2020-10-06:
Hey sorry it took so long to resolve this; was a simple fix, but it came in a month of distractions. I think it should be good now.

closed 2020-10-06 by joeycastillo