joeycastillo/The-Open-Book #39: I Cannot Find Directions

darlingtom, 2020-10-18:
I am probably completely missing the link, so please point me in the right direction.

I do a minor amount of DIY but really enjoy it.  I need basics, though, like a list of parts to buy and step-by-step assembly.  If this exists, please point me in the right direction.  

I'd like to make one and, if successful, make a few more with middle school students for the classroom.  Any help is appreciated.

DCelectronics, 2020-10-18:
I am glad to hear that you would like to make one.

​Depending on your soldering skills, you have to decide if you are going to build the Open Book Feather or the E-Book FeatherWing.

Most of the parts are wide-pitch components, but there are a few fine-pitched components including the flex connector and two chips (ATSAMD51 and the MAX4468 mic preamp). For a simpler build, the E-Book FeatherWing has fewer fine-pitched parts (flex connector), but it requires a [Feather M4](https://www.adafruit.com/product/3857) to function. Below are the choices for ordering boards and parts:

### Open Book Feather A1 Rev 05
[Purchase the Open Book board on Tindie](https://www.tindie.com/products/20599/)
The Open Book bill of materials is [available on Kitspace](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/open-book/).

### E-Book FeatherWing A1 Rev 05
[Purchase the E-Book FeatherWing board on Tindie](https://www.tindie.com/products/19994/)
The E-Book Wing bill of materials is also [available on Kitspace](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/ebook-wing/).

If the board of your interest is not available on Tindie, you can order boards conveniently through one of the three suppliers listed on Kitspace (AISLER, PCBWay and OSHPARK), or download the Gerber files and use the any other board manufacturer of your choice. If you reside in the United States, AISLER also manufactures boards in the [USA](https://aisler.net/blog/aisler-loves-us-manufacturing) with a 2 day lead time for three prototype boards, which thankfully are not purple, and for almost half of what OSHPARK charges for these boards.

Kitspace has choices for parts suppliers (Digikey, Mouser, RS, and Farnel) and makes a shopping cart for you. 
Sometimes, individual parts may be out of stock, and you may have to select alternative part from what is listed in the bill of materials. It is OK to substitute parts with their equivalents. 

That last part is the e-paper display. You will need to order it from [Good Display](https://zh-tw.buyepaper.com/42inch-e-ink-panel-spi-interface-buy-e-paper-display-p0085.html). It ships from overseas though so expect to wait a few weeks. If Good Display is sold out, you can also get the same display from [Waveshare](https://www.waveshare.com/4.2inch-e-paper.htm) for a few bucks more.

### Setup Guide and Documentation
[Setup guide for the E-Book Wing](https://github.com/joeycastillo/The-Open-Book/wiki/E-Book-Wing-Setup-Guide)
[Setup guide and documentation for the Open Book](https://github.com/joeycastillo/The-Open-Book/wiki/Open-Book-Documentation)
[Open Book documentation (PDF)](https://github.com/joeycastillo/The-Open-Book/blob/master/Open%20Book%20Feather/Documentation%20Packet/open-book-printout.pdf)

I hope this response helps you with getting started.

joeycastillo, 2020-10-19:
Thanks @DCelectronics for this response! I'd also add that when I send out the boards from Tindie, I include a documentation packet for each board that includes a blown-up version of the silkscreen, a printed BOM and the booklet that was linked above. All the printouts are available in the folder with the PCB design files: [Open Book documentation packet](https://github.com/joeycastillo/The-Open-Book/tree/master/Open%20Book%20Feather/Documentation%20Packet) and [E-Book Wing documentation packet](https://github.com/joeycastillo/The-Open-Book/tree/master/eBook%20Feather%20Wing/Documentation%20Packet).

In addition, I just added [a 25-minute video demonstrating the build process for the E-Book FeatherWing](https://youtu.be/0Bsh1q2tqaU), which I'll be adding to the README momentarily. This covers hand-assembly, as opposed to using a stencil and solder paste, but this is how I imagine most people will want to assemble the wing. 

Currently no matching video is available for the Open Book, but I think that if you were looking for something to build in a classroom setting, the E-Book Wing is the better board to build (the Open Book has more parts, and a couple are a bit more finicky).

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo