joeycastillo/The-Open-Book #40: Adafruit Compatible Stackable Header for the Feather is too thin and falls out of the

lvw5264, 2020-11-01:
In the BOM, the Top feather header and Bottom Feather header are used to allow the Feather to stack solderlessly on top of the eBook Wing, breadboard style. I would have used the cheaper option of just cutting a Raspberry Pi header in half but I didn't want to risk that.

The feather comes with standard male break away header that is square so it fits snug in the breadboard style header on the eBook Wing. However, these would not allow the FeatherWing wifi to stack on top. Just like these:

https://www.adafruit.com/product/3002

So Adafruit provides a $1 thin pin stackable female header set commonly used in the Arduino for the header to connect to other boards, which is of course, able to stack to female headers like it just fine.

https://www.adafruit.com/product/2830

Unfortunately, these thin pin stackable female headers are too thin compared to the standard male break away header, and easily fall out of the breadboard style one on the eBook Wing, so I could not use them for the purpose of stacking the FeatherWing Wifi chip atop it. Observe the example shown in this product guide to witness the clear difference between the stackable female header Adafruit uses, vs one using square pins just like the standard male break away header:

https://www.rugged-circuits.com/components/8-pin-stacking-header

https://www.rugged-circuits.com/components/6-pin-stacking-header

How did you stack the FeatherWing Wifi chip atop, and with what stackable female headers? Do I have to buy those square ones as above and not flat? it seems to be harder to find, and if so needs to be noted in the guide.

An alternative cheap option is to use longer square header, and then solder the thin pin female stackable header (or a non stackable header if its the last in the stack) to the *bottom* of the FeatherWing instead of the top of the feather. Of course this means that the header is now in an unusual reversed position, but at least Adafruit doesn't solder down the headers so it could work. Did you do that?

joeycastillo, 2020-11-01:
I used the same stacking Feather headers from Adafruit, product 2830, and I have not experienced an issue with the pins being that loose when paired with the headers in the BOM. Can you post a photo of the headers you are using? The ones I have here look like this, and they do fit securely and stay put even when inverted or shaken.

[image]

[image]

lvw5264, 2020-11-04:
I made a video of how the thin pin female stackable header from Adafruit onto the white socket of the BOM just doesn't make a good electrical connection and easily falls out, but I tried to find somewhere convenient to upload it, but I couldn't find it. No clue how yours fits snugly.

I guess the answer is clear to me anyway, I have to use those thicker square female stackable headers as linked above. Do look into it, I'm sure there must be a female stackable header that is properly square on digikey to recommend for the wifi FeatherWIng instead.

joeycastillo, 2020-11-04:
Thanks for the suggestion and I appreciate the report; will definitely search Digikey and see what I can find. Also plan to order some more stacking Feather headers from Adafruit to see if I can reproduce the issue.

DCelectronics, 2020-11-09:
@lvw5264, If you used the BOM parts: Amphenol ICC / FCI 89898-312LF and 89898-316LF, they do not have the typical fork orientation and retention mechanism, and the budget Adafruit headers are too thin for these sockets (the male portion of the stackable Adafruit headers do not have square cross section area). 
The following pictures may help you visualize how the socket engages with the header pins.
[image]
[image]
[image]
These two parts should adders the issue.
https://www.samtec.com/products/ssq-112-03-f-s
https://www.samtec.com/products/ssq-116-03-f-s
[image]

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo