joeycastillo/The-Open-Book #43: Book title and text not showing in MVBook

jxw5883, 2020-11-04:
I'm having this problem where text does not show in MVBook but it does on Quotes demo. It seems that text rendering is not working on menu.h and readbook.h.

I'm using a Feather M4 Express on the E-Book Feather Wing.

[image]

[image]

joeycastillo, 2020-11-04:
That's very odd. Can you confirm that you are using the latest version of both the Babel library and the OpenBook library? And that the BCS solder jumper is closed (I assume it is since it worked with the quotes demo, but just in case). 

[Open Book](https://github.com/joeycastillo/The-Open-Book)

[Babel](https://github.com/joeycastillo/babel)

joeycastillo, 2020-11-09:
@jxw5883 Scratch that — I am able to replicate this issue after updating to the latest version of the Adafruiit_EPD library. It appears to be an issue or incompatibility introduced in version 4.0.0 of the library, though I am unsure as yet whether the problem is in my code or theirs. Either way, I am investigating further and hope to be able to identify a fix shortly. 

For now though, you can address this issue by installing version 3.0.0 of the Adafruit_EPD library in the Arduino Library Manager. This should allow text to once again appear in the MVBook menu and book reading UI. 

[image]

joeycastillo, 2020-11-09:
Got it! It was my bug, fixed in commit b0cbc02. Adafruit_EPD 4.0.0 swapped the order of the white and black colors in their color enum. In MVBook, I had neglected to set the typesetter's text color; this had not caused issues in the past, since its default text color of 0 always corresponded to EPD_BLACK. Now that 0 corresponds to EPD_WHITE, the typesetter was rendering white text on a white background.

Commit b0cbc02 updates MVBook to set the typesetter's text color before rendering either menu text or book text. This fixes the issue.

closed 2020-11-09 by joeycastillo

jxw5883, 2020-11-09:
I can confirm this fixed it. Thanks!