joeycastillo/The-Open-Book #44: Open Book vs Open Book Feather wing

wa2mze, 2020-11-11:
I just ordered two displays ("First rule of government spending, why buy one when you can have two at twice the price?"), couldn't justify spending as much on the shipping as the hardware, the second display was $1 cheaper and free shipping (sorta).  I will order a board once I figure out which one (might get both, see above rule :-) ). 
Questions:
1:  Why does the "wing board" have an added serial RAM chip that the "full" board does not?
2: Can I use the SAMD51J20 (twice the memory) instead of the 'J19? (I've got two of them in my parts bin).
3: Will there be a new hardware spin on either board soon (like to add that 'missing' ram chip to the "full" board)?
My SMT solder skills are probably up to the task, I've soldered 0.5mm QFTPs many times with good results.  The first time I tried to use paste, a stencil and hot air it was a disaster! (couldn't remove solder bridges UNDER the legs, removed the chip and lost it when it fell on the floor!).  I'll stick with drag soldering by hand and have wick handy!

joeycastillo, 2020-11-12:
1. The wing has the SRAM chip, at least in theory, for pairing with Feathers with less RAM (like the old school ATmega32u4 Feathers). It's actually not currently used by the Open Book software library, because the SAMD51 has enough RAM to buffer the whole display (and the SRAM chip doesn't seem to play nicely with some of the partial update stuff I'm doing). 

2. You should be able to use the SAMD51J20 as it is pin compatible, but you might need to hack around with the software support libraries. The Arduino library specifies the SAMD51J19 in the board definition, and has Flash sizes tailored to that chip. The CircuitPython build also specifies the SAMD51J19, but I don't think you'd get any benefit there as the CircuitPython runtime fits comfortably in 512 KB; all your user code will live on the 2MB Flash chip anyway. 

3. I'm not planning a new revision of either board anytime soon, and in particular there is no plan to add the SRAM chip to the Open Book, since the SAMD51 has enough RAM for the screen. The only reason I would consider another revision to the E-Book Wing is that it uses a different style of button for lock and reset than the Open Book, but I'm hesitant to make yet another revision at this time (and both kinds of button work just fine).

DCelectronics, 2020-11-27:
@wa2mze These are great questions.

1. I had the same question during the design conversion to KiCad, but I figured some of the parts can be optionally installed/populated on the board, depending on the desired features (such as connectors for peripherals, QSPI Flash, etc). For example, if you don't care for CircuitPython, you can probably skip installing the QSPI Flash chip on the board.

2. For SAMD51J20, you may consider looking at the libraries developed for [Sparkfun DEV-14713](https://www.sparkfun.com/products/14713) and modify the Arduino SAMD core for the Oddly Specific Objects. One thing with the Open Book Featherwing is that you are not limited to the Adafruit products. You can use the [Sparkfun DEV-14713](https://www.sparkfun.com/products/14713), or get the [Teensy 3.x Feather Adapter](https://www.adafruit.com/product/3200) and try out the [Teensy 4.0](https://www.pjrc.com/store/teensy40.html), which is almost 11 times faster than the SAMD51J19, and that would _ruffle a lot of feathers_.

3. I am thinking of making a variant board. There is plenty of space on the board to add optional features such as RTC (DS3231), WIFI (ESP32), accelerometer, etc. By the way RTC is possible on the Feather M4 Express. Here is the [library](https://github.com/ericbaril72/RTCZero/tree/SAMD51_E54/src) that supports SAMD51 WITH CRYSTAL. If you get issues that the library is old just install the latest from Arduino library manager (current version 1.6.0) and overwrite only the [RTCZero.cpp](https://github.com/ericbaril72/RTCZero/blob/SAMD51_E54/src/RTCZero.cpp). I tried this on the Feather M4, Metro M4, and Grand Central M4, and it worked on all three different boards. This modified library did not work on ItsyBitsy M4 because it is crystal-less. Similarly, I am not expecting this library to work on the Open Book (because it is crystal-less too). It may be possible to add support for crystal-less SAMD51 (SAMD21 crystal-less is supported). So, why bother with separate RTC chip if the SAMD51 can do that? Well the  DS3231is super accurate clock with temperature compensated encapsulated crystal.

wa2mze, 2020-11-29:
I hadn't thought about the 'other' feathers that could be used with the Open Book Feather Wing version.  Adafruit has an STM32F405 board with 1mb flash and 192kb ram, TWO D/A converters (Stereo audio), and a micro SD socket.  Then there is also the ESP32 with BT and WiFi built in.  Rewiring the headphone socket would be required to use the second D/A, don't know if that would be possible.  Can the software support the on board SD socket (eliminating the need to solder one onto the board)?

DCelectronics, 2020-11-30:
@wa2mze I think you can use the STM32F405 on board micro SD socket. I suggest you read issue #31.
I also recommand to give it a try to solder the microSD socket onto the board if you have a fine tip soldering iron (watch this [video](https://youtu.be/0Bsh1q2tqaU?t=878) to see how Joey is doing it). After watching the video, I concluded that soldering the micro SD is not as hard as I originally thought.

wa2mze, 2020-11-30:
I've watched the video.  It doesn't look too bad soldering the SD socket, assuming you have the 'right' socket.  The one Adafruit sells has the card detect pin, but I don't know if the access holes match the one in the video.  I've read #31, sounds like a jumper will be required.  It looks like I can wire the A1 pin to the headphone jack for stereo, but will have to 'dead bug' a second attenuator network for the extra channel.  I would leave out the connectors for A1 and A2 and only attach the external I2C connector (well MAYBE A2 if I can attach a microphone to that).   The Teensy 4.1 also looks interesting.  While longer than the 4.0, it might work on the Feather adapter if the pinout for the 4.0 portion is compatible.  The 4.1 has a FULL micro SD socket connection, not just a simple SPI, and Teensduino has a library that supports it.

DCelectronics, 2020-12-01:
The Adafruit [micro SD socket](https://www.adafruit.com/product/1660) appears to be the same part as the one used on the Open Book, [Würth Elektronik 693071010811](https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/693071010811/3124583?s=N4IgTCBcDaIOwGYwFoEA4CMBOZHkDkAREAXQF8g), if you look at the [datasheet](https://www.we-online.de/katalog/datasheet/693071010811.pdf).
The Teensy 4.1 may not work on the [Teensy 3x Feather adapter](https://www.adafruit.com/product/3200) without some creative hacking (maybe tall headers). Teensy 4.0 and 4.1 support RTC out of the box (Teensy 3.2 requires soldering a [32.768 kHz crystal](https://www.pjrc.com/teensy/td_libs_Time.html#teensy3)), and you can proceed without a DS3231.
One thing I don't like about a single set of feather headers is that every added featherwing stacks up. Maybe have the option to solder up to four sets of feather headers on the board would allow to have multiple featherwings spread out. I am also thinking the option to have surface-mount single-row male pin headers, which are easy to solder on a board (just use female sockets across for spacing) and solder on the male pins a Teensy 4.1. The babel chip (SPI Flash) [can be soldered](https://forum.pjrc.com/teensy41_memory.jpg) directly on the Teensy 4.1.
The possibilities are endless. Open Book... Open Mind...

DCelectronics, 2020-12-01:
The Adafruit micro SD socket appears to be the same part as the one used on the Open Book, Würth Elektronik 693071010811, if you look at the datasheet.
The Teensy 4.1 may not work on the Teensy 3x Feather adapter without some creative hacking (maybe tall headers). Teensy 4.0 and 4.1 support RTC out of the box (Teensy 3.2 requires soldering a 32.768 kHz crystal), and you can proceed without a DS3231.
One thing I don't like about a single set of feather headers is that every added featherwing stacks up. Maybe have the option to solder up to four sets of feather headers on the board would allow to have multiple featherwings spread out. I am also thinking the option to have surface-mount single-row male pin headers, which are easy to solder on a board (just use female sockets across for spacing) and solder on the male pins a Teensy 4.1.
The possibilities are endless. Open Book... Open Mind...

wa2mze, 2020-12-01:
I'm still on the fence as to which way to go, the OpenBook feather wing, or the OpenBook feather.  I'll probably buy one of each as I did buy two of the screens.  Both ways have their pluses, the wing offers the possibility of trying several different processor options, and the feather allows easier stacking of feather wings to add functionality in a thinner device.  The wing is probably easier to build, so I would go that way first.  BTW DCelectronics, if you do layout a custom version, take a look at Adafruit's I2S headphone amp breakout board, that would be the way to add Stereo audio output to either the feather or the wing board (16 or 24 bit instead of 12) for HiFi Stereo.

DCelectronics, 2020-12-01:
I see your dilemma, and the OpenBook featherwing gives you the freedom to chose a microcontroller that you are comfortable with. 
The only thing with the OpenBook feather is that you are limited to the SAMD51J19 microcontroller (with no 32.768 kHz crystal) and make sure you have the means (Atmel-ICE, Arduino Zero EDGB, Segger J-Link, etc) to write to the bootloader on the blank SAMD51J19 chip. The most challenging part to install on either OpenBook featherwing and OpenBook feather is the display connector.
While thinking more into this, its possible to add more featherwings by using flat arrangement with [Adafruit product 3417](https://www.adafruit.com/product/3417). (It gets a bit tricky with the battery connector. You may have to solder a 2-pin JST connector somewhere on the prototype area and wire it to the feather battery pins).
[image]
[image]

Looking into the I2S design that uses the NXP UDA1334A, it appears to me that the chip is reaching the end of life (obsolete).

Going back to Teensy 4.1, the Babel SPI Flash chip can be soldered onto the back side of the [Teensy 4.1](https://forum.pjrc.com/teensy41_2.jpg).

wa2mze, 2020-12-01:
Well, I just ordered one of each board as the cost won't  break the bank.  For the 'feather' board, I'll probably mod the source to accept the 'J20' chip with double the memory.  I doubt that this will be a large problem.  I had already ordered two of the displays, and they showed up yesterday in the mail.  I like how they were shipped with the "OK" from the factory test still preset on the display.  Non volatile display.  Nice.  So I can eventually build one of each.  I didn't know the UDA1334 was near EOL.  Guess Adafruit won't be selling this much longer, unless they made a last time purchase to last for awhile.  The Tripler board is a nice idea.  BTW Adafruit sells the SMT feather connectors.

DCelectronics, 2020-12-02:
The fact that the [UDA1334 has reached end of life](https://www.nxp.com/part/UDA1334ATS#/) is not the end of the world. It can be sourced from non-standard suppliers. 
I am glad to hear that your displays arrived OK. Even though the resolution is 400x300 at 120 dpi, it has a nice dot-matrix retro look without taxing the microcontroller ram. I appreciate the advantages of the microcontrollers running C/C++ over the microprocessors running Linux, and therefore I appreciate the Open Book design.

joeycastillo, 2020-12-02:
FYI, in case it helps with any ideas for custom designs, I am working on a new variant at the moment. It's not intended to replace the existing design, but rather to see if I can make something that is cheaper to manufacture. It has a different layout, and thoughts about FeatherWings are a big part of it: 

[image]

You'll notice the Feather headers are relocated to the bottom left, away from the USB port. The reasoning:

1. For most FeatherWings with ports, the connectors are on the side where the headers are even. For example, [this Ethernet wing](https://www.adafruit.com/product/3201), [this power relay](https://www.adafruit.com/product/3191) and even this [forthcoming CAN Bus FeatherWing](https://youtu.be/sIiG97f3VbQ?t=1228), all have useful ports on that side of the board. Moving this side of the Feather connector to the edge of the device seems to make the most sense, since the port will be available on the "outside" of the device.
2. By keeping the wing area devoid of other parts, one could omit the Feather headers and solder a wing directly on to the board. You'd either have to sand off the edges of an existing wing (or design a wing with castellations), and it would only work for wings with no protrusions on the bottom. But if starting from scratch, it strikes me as a way to design different modules that could permanently augment the design for specific use cases, while retaining a slimmer profile.

I have not built this variant yet — I'll push the design files up once I do and I have a chance to test it — but sharing the thought process now in case it sparks further ideas!

DCelectronics, 2020-12-02:
I like this with an [ESP32-WROVER-E (16MB Flash, 8MB SRAM)](https://www.digikey.com/en/products/detail/espressif-systems/ESP32-WROVER-E-16MB/11613135) module installed:)
I also like the larger battery area. 

I am thinking to: 
* Mill a notch on the edge and have the screen flex cable folding over the the PCB (similar to the [inkplate](https://www.crowdsupply.com/img/040a/inkplate-6-angle-01_png_project-body.jpg) but not as pronounced). The benefit is that the screen glass substrate would lay flat on the PCB with the flex cable having a large bend radius, and easier screen assembly. This forces me to move the STEMMA connectors at the bottom of the PCB, and I can live with that.
* Stick to the micro USB because it has a better mechanical strength on the PCB than the USB-C while there are no USB-C benefits on this design (no USB 3 speed, no higher than 5V for fast charging).
* Follow the Espressif hardware design guidelines for keep-out areas as much as possible [(Section 3.1.2)](https://espressif.com/sites/default/files/documentation/esp32_hardware_design_guidelines_en.pdf).
* Use the typical FTDI pin arrangement for the UART (with the Vcc pin connected to the 5V since most USB-UART cables have 3V3 TTL and 5V Vcc). If not having onboard USB-UART chip, I would include the [auto-program circuit](https://www.smart-prototyping.com/image/data/2_components/development_board/101839%20NodMCU-32S/1.png) going out to the UART header with the optional IO0 and #EN buttons.
* Change the micro SD socket to an easy one to handsolder [(Molex 5031821852)](https://www.molex.com/molex/products/part-detail/memory_card_socket/5031821852).

wa2mze, 2020-12-02:
The ESP32 feather was one of the boards I'd consider plugging into the OpenBook featherwing (is this module the one that 86's BT?)
Running "raw" C/C++ has advantages of more control of power consumption, low "bloat", and quick responsiveness.  However, Linux does supply quite a bit of the "backbone" required such as file systems, drivers, and libraries.  The Raspberry Pi Compute Module might be a suitable platform for a full feature Open Book, but would demand a custom Linux image that only contained the necessary drivers, daemons, and utilities.  It should not have any 'desktop' such as Gnome or KDE, just a stripped bare X11 window manager customized to run an epaper display.  In fact it would be a mostly headless interface, with the ebook app AS the shell, with extensions for a file browser, and perhaps an FTP client to download new books.  I'd expect such an expanded version to have a larger display with touch screen, perhaps something in the range of 6-8".  GooDisplay has some in this size range, though a separate touch overlay might have to be laminated on after the fact. 
I know these ideas are boardering on an open source iPad, but the update rate of eink displays isn't really suitable for a web browser, and ideal only for 'read only' document viewers.  I'm at the age where 'large font', high contrast displays are more comfortable to read (old eyes) and the 4.5" display might be marginal for me.

DCelectronics, 2020-12-02:
@wa2mze While the SAMD51 supports Native mode SDHC/MMC the TQFP64 package brings out to its pins only the SDHC0 data lines [(See Datasheet pages 32 and 33)](https://www.mouser.com/datasheet/2/268/60001507A-1130176.pdf). SDHC0 is used on the current Open Book design for the QSPI Flash chip.  There is the choice to remove the QSPI Flash chip to wire the SD socket to SDHC0 and kiss CircuitPython good-bye, or change the SAMD51 package to the 100 or 128 pin variant in order to wire the SD socket to SDHC1.

wa2mze, 2020-12-02:
I assume that the ereader software doesn't need either the QSPI flash or Circuit Python.  OTHO, in this application, the SDHC access of the SD might not make that much difference in performance over SPI access.  I'll keep it in mind as a possible experiment.  I wasn't going to mount the QSPI flash if it wasn't needed for the ereader application.

DCelectronics, 2020-12-03:
@wa2mze There is an option to use [CircuitPython](https://github.com/joeycastillo/The-Open-Book/tree/master/CircuitPython) on the Open Book for beginners. I looked at the Grand Central M4 design, and I saw that 1 data line SPI is used for the micro SD socket. The current Arduino libraries are using the SPI (1 data line), and for the SAMD51 native mode SDHC/MMC the AtmelStart is a place to start. I have used 4 data lines for micro SD on an STM32F7xx hardware design with fast image transferring requirements, but someone else had the pleasure to work on the firmware. 

@joeycastillo Are you routing the native USB pins of the ESP32-S2 to the USB-C connector? I couldn't tell from the posted pictures.

wa2mze, 2020-12-08:
Yet another feather board is the Adafruit N$F52840 Sense (https://www.adafruit.com/product/4516).  M4F processor with 1M flash and 256K ram, and a ton of sensors. Also has 2M of QSPI flash.  No DAC though, so headphone jack won't work.

DCelectronics, 2020-12-09:
The Adafruit product [4516](https://www.adafruit.com/product/4516) you mentioned is a step up from the barebone [Adafruit Feather nRF52840 Express](https://www.adafruit.com/product/4062). I am only having a Clue (pun intended) with nRF52840.
I acquired a Teensy 4.1, and it is a "beast". The SD is using 4 data lines. I like that I can add an SPI Flash chip at the bottom of the PCB.
For audio, I tried yesterday the [Adafruit Music Maker FeatherWing - MP3 OGG WAV MIDI Synth Player](https://www.adafruit.com/product/3357), and I was [streaming internet radio](https://learn.adafruit.com/adabox004/internet-radio) (128kbps mp3) on ESP8266. It is a great add on for your audio needs [(such as mp3s)](https://learn.adafruit.com/adabox004/playing-your-first-mp3) without taxing your microcnotroller.

wa2mze, 2020-12-09:
The adafruit 4481 https://www.adafruit.com/product/4481 is cheaper and 
has 2m of QSPI flash. But it's and ItsyBitsy, not a feather.  Maybe 
someone should make an ItsyBitsy adapter board. The ItsyBitsy boards 
leave off the Lipo charger circuits, are a bit narrower (600 mill 
connector row spacing vs 800mil (board width 0.7" vs 0.9").  The 
ItsyBitsy boards can actually have more i/o pins than the Feathers.

DCelectronics, 2020-12-10:
A compatible board to the existing SAMD51 Feather is the ItsyBitsy M4, and it is cheaper than the ItsyBitsy nRF53840. The ItsyBitsy M4 features the 48 pin SAMD51G19A (instead of the 64 pin SAMD51J19A found on the Feather M4) and what is missing from the big Feather is the I2S. 
Making an adapter from ItsyBitsy to Feather is feasible by taking the design of the Teensy 3.x feather adapter, change the nets on the schematic and reroute the board (leave the footprints intact and in place as the ItsyBitsy and Teensy 3.x / 4.0 have the same outer pin footprint). 
Why making an adapter for ItsyBitsy to Feather if a lot of the Adafruit boards are offered on the both from factors (ItsyBitsy and Feather)?
With the Teensy, it is different case, because no Teensy is offered on a Feather form factor.

wa2mze, 2020-12-16:
So I just ordered the parts to build the OpenBook Featherwing.  I'll leave the OpenBook with on board processor for later.  Adafruit had the SMT feather connectors.  I also ordered their "triple-ler" wing adapter which can be added later to add other wings (like for wifi).  I have three possible feather CPU's, the SAMD51, the ST32F405, and the ESP32.  (and now there is a SAME51 too, but I don't think the CAN bus would add much here).  I'll probably try configuring the software for all three of these and see which works best.

joeycastillo, 2020-12-16:
FYI, there was an error on the BOM for the wing... wrong MicroSD slot per conversation in #47. I thought I committed this change the other night but [here's the diff](https://github.com/joeycastillo/The-Open-Book/commit/12a805342cbac59305abd619eb2bfacb47c4c757) and the [correct part on DigiKey](https://www.digikey.com/en/products/detail/würth-elektronik/693071010811/3124583). You can also use [this part from Adafruit](https://www.adafruit.com/product/1660); the footprint is identical for both.

wa2mze, 2020-12-16:
I ordered the Micro SD socket from Adafruit, so I think I'm good here.

wa2mze, 2020-12-16:
I like the idea of the ESP32 based board.   That CPU module will also be easier to solder than a QFTP-64!

DCelectronics, 2020-12-17:
@wa2mze I got (for another application) a SAME51, unfortunately form Adafruit, and I discovered that it is lacking support. So, I am trying to use this [library](https://github.com/Longan-Labs/CANBed_M4_Arduino_Lib) for this [product](https://www.longan-labs.cc/1030013.html), but I still have to adjust a few things.  
As far as the ESP32 module being soldered directly on the PCB, "Soldering Pad 39 to the Ground is not necessary for a satisfactory thermal performance." is a nice note on the [datasheet](https://www.mouser.com/datasheet/2/891/esp32-wroom-32d_esp32-wroom-32u_datasheet_en-1365844.pdf). 
I don't have a reflow oven, yet, and I had this idea, to place a through hole at the center of pad 39, and melt the solder in the hole from the back side with the soldering iron. It this case, I don't have to solder the ESP32 pad 39.

wa2mze, 2020-12-21:
Adafruit order arrived this morning.  So I should now have all the parts required for the OpenBook Featherwing.  Guess I'll start assembly with the small SMD parts (resistors, capacitors, transistors), and then move on to the more challenging ones (IC's and connectors).

That ESP32 module has 3 or 4 edge pads connected to the same "ground" as pad 39, so the bottom pad isn't needed for power and signal ground common.  Since it doesn't seem to be needed for heat sinking, I guess you can just ignore it, if your board can provide good enough shielding via ground pours.

joeycastillo, 2020-12-21:
> I don't have a reflow oven, yet, and I had this idea, to place a through hole at the center of pad 39, and melt the solder in the hole from the back side with the soldering iron. It this case, I don't have to solder the ESP32 pad 39.

I didn’t see that note, but FWIW, I had success putting some low temperature Maker Paste on the pad, and blowing hot air on the vias from below. In the future, though, I may not bother if it’s not strictly necessary.

wa2mze, 2020-12-21:
I just looked at the pinout diagram for the module and noted the other ground connections.  Since they all have the same netname, I expect them to be all tied together.   
Putting a via hole connected to ground directly under the pad and then just melting solder into the hole from the back side of the PCB with a small soldering ion tip should work fine.  Same for solder paste and hot air.  Smear a bit of flux on the back of the module before attaching it to the PCB.  I've attached the bottom contact of an AD9951 to a PC board that way (soldering iron), it works.  If you are paranoid, you could also stick a small bit of wire in the hole to insure electrical contact with the chip pad.  I suspect that pad is mostly for mechanical hold.

DCelectronics, 2020-12-22:
> I didn’t see that note, but FWIW, I had success putting some low temperature Maker Paste on the pad, and blowing hot air on the vias from below. In the future, though, I may not bother if it’s not strictly necessary.

@joeycastillo I am sorry for not providing more details. It is on page 16 of the [datasheet](https://www.mouser.com/datasheet/2/891/esp32-wroom-32d_esp32-wroom-32u_datasheet_en-1365844.pdf) (or page 21 of 28). 

[image]

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.