joeycastillo/The-Open-Book #45: Bookbinder script issues

wa2mze, 2020-12-03:
I tried running bookbinder.py and got these error messages:

ken@ZenPenguin:~/Arduino/projects/Ereader/The-Open-Book/Examples/Open_Book_MVBook$ ./bookbinder.py 
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/408.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/408.
import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/408.
./bookbinder.py: line 5: class: command not found
./bookbinder.py: line 6: syntax error near unexpected token `('
./bookbinder.py: line 6: `    def __init__(self, offset, length):'

I think this is the version of python I have installed: 
ken@ZenPenguin:~/Arduino/projects/Ereader/The-Open-Book/Examples/Open_Book_MVBook$ python
Python 2.7.18 (default, Aug  4 2020, 11:16:42) 
[GCC 9.3.0] on linux2

I'm not sure how to run the bookbinder script, (no instructions provided) IE: are any arguments required, or which version of python is required.

joeycastillo, 2020-12-03:
It’s Python 3, you should be able to just run `python3 bookbinder.py`. It’s an interactive program that prompts you with actions to take; you’ll be asked for the title of the work, then the author, and a file name (without the extension). It will take this input and generate a file with the same name and the extension “OBK”. 

After that, the program will ask for another title in case you have more works to package; if you don’t, you can exit the program with Control+C or Control+D.

wa2mze, 2020-12-03:
Thanks!
No error when I run it with python3 (which I seem to have installed on my Kubuntu system).  I would have thought that the python script should have had a line specifying which python to run if the script was called out in bash with a ./bookbinder.py to launch it (assuming the script was executable).  Bash scripts can be so launched under the shell.  This migration from python to python2 to python3 has been a PITA.

Can I specify the full path for the file, or do I have to run the script from the directory where the files reside?

nfraprado, 2020-12-03:
In the first case the script was probably being run as bash script, and in the second case, with python2, but as joey mentioned, it should be executed with python3.

This can be made default, so that you can execute it like `./bookbinder.py` without errors, if you add a `#!/bin/python3` as the first line of the file. This change should probably be made on the repo as to avoid others to stumble into this error as well.

wa2mze, 2020-12-03:
YES, that's the bash like line I was thinking of.  It might also have been a good idea to adopt a .py3 extension for the script (with .py2 and .py as well) to indicate which python the script is for, since the differences are enough to really make it a different dialect of python!

nfraprado, 2020-12-03:
Well, I don't really agree about changing the extension, as I haven't really come across `py3` extensions on the different python projects I have worked with. Also, python2 finally reached end of life this year, so it's all python 3 moving forward.

What I do think would be valuable is to have that shebang for python3 in the file, as well as instructing the user to use python 3 in the documentation.

joeycastillo, 2020-12-03:
I should definitely add an interpreter directive on line 1, that's an easy fix and something I overlooked. 

> Can I specify the full path for the file

I don't think so, you should probably run the script from the directory with the files you are converting. It's an admittedly fragile script, because the OBK format is not meant to be a permanent solution. My goal is to allow folks to drop .txt files on an SD card (presumably with some YAML-like front matter for title and author) and have this processing happen on the device.

wa2mze, 2020-12-03:
Thanks Joey.  I agree, add the #! line to the script (and the other python scripts in the project) to indicate python3.  While python2 may be EOL, most distros still distribute it because they are too lazy to convert all of their internal scripts to Python 3.  I suppose I could add a line to my .bashrc to redirect all python calls to python3, but the #! in the script would be safer.  Also add something to the documentation (readme file) about python3.

Eventually, the MVbook app should include the ability to process ebook and pdf files.  This would be easy in a turnkey Linux system as all the necessary libs for this are there.  For a 'naked' C/C++ implementation under Arduino the Linux libs could be ported.  I might eventually look into that myself and see if I can figure it out, but I'd first have to get familiar with the way that text is currently formatted and written to the epaper display.  Is there a sort of API in the app/libs for this?

As soon as my boards arrive, I'll look at making up a BOM to buy the parts.  Some will be from Adafruit, the rest from Digikey and or Mouser. (Mouser is sometimes a bit cheaper that Digikey on certain items, but if you snail mail Digikey a live check for payment they will grant you free shipping on your order, no matter how large or small).   I wonder if the ESP32 feather will work (4MB flash, 512K ram, 2 D/A, WiFi and BT.  But some I/O pins are INPUT or OUTPUT ONLY!)

joeycastillo, 2020-12-03:
> I wonder if the ESP32 feather will work

It has been a long time, but I did test with the ESP32 Feather way back at the beginning, and found that a baseline level of functionality did work (at the very least, updating the screen).

> I'd first have to get familiar with the way that text is currently formatted and written to the epaper display. Is there a sort of API in the app/libs for this?

The Open Book uses an 8x16 (sometimes 16x16) bitmap font called GNU Unifont. It does this so that it can support just about all the world's languages out of the box. I do this with a support library called [Babel](https://github.com/joeycastillo/babel), which cooks all the data into a blob called `babel.bin` that allows you to quickly look up any character in the Unicode basic multilingual plane. 

Documentation is mostly inline: the [BabelTypesetter](https://github.com/joeycastillo/babel/blob/master/src/BabelTypesetter.h) describes how the text makes it on the screen, and the [BabelDevice](https://github.com/joeycastillo/babel/blob/master/src/BabelDevice.h) describes at a high level how to get Unicode character information and Unifont glyphs out of some hardware device with babel.bin on it (in the case of the Open Book, [a 2MB SPI Flash chip](https://github.com/joeycastillo/babel/blob/master/src/BabelSPIFlash.cpp).

commit 703861d9 (2020-12-12): examples: add shebang to bookbinder.py (fixes #45)
closed 2020-12-12 by joeycastillo