joeycastillo/The-Open-Book #47: Incorrect FPC Connector P/N for the ePaper

DCelectronics, 2020-12-11:
The FPC (flexible printed circuit) connector specified in the BOM (for connecting the ePaper flexible cable) is [Amphenol-ICC P/N: SFV24R-4STBE1HLF](https://www.amphenol-icc.com/fpc-ffc-sfv24r4stbe1hlf.html).
According to the [application specification](https://cdn.amphenol-icc.com/media/wysiwyg/files/documentation/bjm-sfv102.pdf) (see page 3), the selected connector is designed for use with FPCs that have cut-out sections.
[image]
These are some [connecotrs](https://www.digikey.com/en/products/filter/ffc-fpc-flat-flexible-connectors/399?s=N4IgjCBcoGwJxVAYygMwIYBsDOBTANCAPZQDaIALAAxwUBMVIAuoQA4AuUIAyuwE4BLAHYBzEAF9CADimIQKSBhwFiZcAFYwcOghYgOXXoNETpCaPLRY8hEpHLq6YOszadIPfsLGTwYMLIWCko2qvbgUgDM6giEAZEwFCBxcOowMK767p7GPnGRcBBBVsq2atRScInJlFSVcADsmQYeRt6m4A0UTcWK1ip25JHD9Lpuhl4mvlqVcsH9ZeGMei057b4AtC69-ACuA2rqzdkS4uJAA) that could work with the ePaper FPC, but further verification of the footprint is required before selecting one.
The FPC connector supplied with the ePaper screen by [Good Display](https://zh-tw.buyepaper.com/42inch-e-ink-panel-spi-interface-buy-e-paper-display-p0085.html) appears to be an [Omron P/N: XF3M-2425-1B](https://www.mouser.com/ProductDetail/Omron-Electronics/XF3M-2425-1B?qs=vLWxofP3U2xSl4%2FL5wv8NQ%3D%3D)

joeycastillo, 2020-12-12:
Oh wow. This is completely correct, and a mistake that dates back to the first boards I made. I had noticed the tight fit, but since all the contacts worked, and I didn't know any better way back then, it remained on the BOM with a note suggesting to use the one Good Display included. 

The good news is that the connector in the BOM is functional, though it sort of bends the flex connector on either side. Once the display is taped down though, I have had no issues with the performance of the screen. 

[image]

I will order some of the connectors in the search you provided and make an update to the BOM next week.

DCelectronics, 2020-12-13:
The FPC is about 12mils (0.30mm), or the thickness of three sheets of a typical printer paper, and it gets wedged there (thus the tight fit).

Attached is a zip file with screenshots of some FPC connectors that I could find their 3D model for checking them on the footprint. Couple of the connectors (0524352471 and SFV24R-2STBE1HLF) have their supporting tabs partially landing outside the area of the footprint pads. The rest seem OK. The filenames in the zip file match the manufacturer part number.
[FPC_Connectors.zip](https://github.com/joeycastillo/The-Open-Book/files/5684204/FPC_Connectors.zip)

A separate issue with the BOM is that the micro SD part number for the eBook Wing is [Molex 5025700893](https://www.digikey.com/en/products/detail/molex/5025700893/1866792) and it should have been [Würth Elektronik 693071010811](https://www.digikey.com/en/products/detail/w%C3%BCrth-elektronik/693071010811/3124583?s=N4IgTCBcDaIOwGYwFoEA4CMBOZHkDkAREAXQF8g) or [Molex 0473340001](https://www.digikey.com/en/products/detail/molex/0473340001/2404955?s=N4IgTCBcDaIOoFkDMYAcqAiBpASgWgDkMQBdAXyA). The [Molex 5025700893](https://www.digikey.com/en/products/detail/molex/5025700893/1866792) may work, but I don't know this for sure.

joeycastillo, 2020-12-13:
> A separate issue with the BOM is that the micro SD part number for the eBook Wing is Molex 5025700893 and it should have been Würth Elektronik 693071010811 or Molex 0473340001. The Molex 5025700893 may work, but I don't know this for sure.

Oh no, that is completely wrong. A few months ago I ordered several microSD connectors to check compatibility, and that one specifically had a footprint that didn't match. My intent was to list the Würth Elektronik part for both devices; I don't know how I got the wires crossed but will fix it now.

DCelectronics, 2020-12-17:
@joeycastillo @kasbah How can we update the [BOM changes](https://github.com/joeycastillo/The-Open-Book/commit/12a805342cbac59305abd619eb2bfacb47c4c757) on the [Kitspace](https://kitspace.org/boards/github.com/joeycastillo/the-open-book/ebook-wing/)?
@joeycastillo Joey, did you have a chance to look at the [FPC_Connectors.zip](https://github.com/joeycastillo/The-Open-Book/files/5684204/FPC_Connectors.zip) I posted for the FPC connector?
Except Molex 0524352471, all others ate good choices. My favourite choices are the [Hirose FH12A-24S-0.5SH(55)](https://www.digikey.com/en/products/detail/hirose-electric-co-ltd/FH12A-24S-0-5SH-55/966386) and [Omron XF3M-2425-1B](https://www.mouser.com/ProductDetail/Omron-Electronics/XF3M-2425-1B?qs=vLWxofP3U2xSl4%2FL5wv8NQ%3D%3D).

joeycastillo, 2020-12-17:
I think that Kitspace automatically updated; I'm seeing the Würth Elektronik card slot listed there now. 

I did look at the zip file, and I was hoping to order some of these parts just to see how they fit with the screen and the footprint before updating, but if you feel certain that these parts will work, perhaps it's worth updating it now. Between the two I would lean towards the Hirose connector, just because it's the one that's available on Digikey and it makes Digikey a one-stop cart for everything except the display.

DCelectronics, 2020-12-17:
I guess there was a time delay in updating the Kitspace when I looked. Great! The SD socket issue is resolved.

I agree that the Hirose FH12A-24S-0.5SH(55) connector is available on both [Digikey](https://www.digikey.com/en/products/detail/hirose-electric-co-ltd/FH12A-24S-0-5SH-55/966386) and [Mouser](https://www.mouser.com/ProductDetail/Hirose-Connector/FH12A-24S-05SH55/?qs=%2Fha2pyFaduhBL9Ra9C9JJo9wVmxbvHTxvQYdc0zFc5HpowawelgM75Z1xbHioLdK).

I looked at the connector datasheet, the e-paper screen datasheet and the connector footprint in the KiCad files. I concluded that the Hirose FH12A-24S-0.5SH(55) connector is suitable for the epaper screen FPC, and the connector footprint in the KiCad files meets (bounds) the specifications found on the connector datasheet. You can independently check that, and proceed with this part number.  

These part numbers: SFV24R-2STBE1HLF, F32Q-1A7H1-11024, and 0545502471 are also available on both Digikey and Mouser, but the footprint recommendations (for their support pads) I found on their datasheets are not met in the existing layout footprint for the connector - these parts would work, but not optimally.

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.