joeycastillo/The-Open-Book #49: E-Paper Display

markgibb02, 2021-02-18:
The BOM calls for the GDEW042T2 display from Good Display. As far as I can see, this has been sold out for months, with no sign of coming back in stock. Is there a direct replacement display which can be used without changing hardware and code?

joeycastillo, 2021-02-19:
I hear you and totally feel your frustration. I’ve been in touch with the manufacturer and the fact is, they’re experiencing long lead times for the glass necessary to make the display. I don’t know if it’s pandemic related, but given how much it has turned the world upside down, I wouldn’t be surprised. The expectation is that they will be able to resume production around April. 

At this time there really isn’t another display that’s either mechanically or code compatible with this design, and moreover none that I’ve seen that can hit the same price point that would keep the project inexpensive and accessible. Given all that, my instinct is to just wait out this supply crunch and hope things get better as the year progresses.

MixedMatched, 2021-03-18:
It looks like the display is [back in stock](https://buy-lcd.com/products/42inch-e-inkanel-spi-interface-buy-eaper-display).

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo