joeycastillo/The-Open-Book #56: Dead project?

mcmurphy8097234789, 2022-04-24:
This project has not seen a commit since 2021. Is the project dead? I hope not, it's really cool

nerdguy1338, 2022-04-24:
It's only been a year.

The world's a little crazy right now, give it time he'll come back.

joeycastillo, 2022-04-25:
TL;DR: Not dead. First version had issues as it was my first real electronics project, and last year I chose to focus on simpler projects that could teach me lessons that will serve the book well when I return to it. (also chip shortage has been hard)

It's funny you opened this issue this weekend: I actually spent some time on Friday assembling the ESP32-S2 version of the Open Book (`OSO-BOOK-B1-01`). It's still a project I want to see to completion. But the Open Book that I designed in 2020 was essentially the first electronic gadget I'd designed, and as such it suffered from several issues. First, it's way too complex and has far too many parts. This makes the BOM cost exorbitant, to the point that I'd never be able to manufacture it affordably. I also neglected to include a keepout area for a reasonably sized battery. To make matters worse I also didn't understand low power design, so the book's quiescent current draw was too high to achieve weeks-long battery life on the battery that did fit there. That's to say nothing of the firmware: it's an Arduino sketch, which is a shaky foundation that doesn't lend itself well to community contributions.

These are just a few of the issues, but the common thread to all of them is that I didn't know any better when I started, and I learned these lessons over time. At many points I've come up with changes that could make the Open Book a better gadget; you can check out [these notes](https://twitter.com/josecastillo/status/1375901032660733952) for an example of where my thinking was at this time last year. Still: even at that time I knew there were lessons I still had to learn, especially around manufacturing at a larger scale and delivering firmware that a community can build on, and I knew that even these slimmed down versions were too complex to be the gadgets I cut my teeth on. (there were also shortages of everything from chips to glass for the e-paper display)

ANYWAY. I made a decision last year to focus instead on learning these lessons by building and manufacturing [Sensor Watch](https://github.com/joeycastillo/Sensor-Watch), and I think I'm doing well: I'm figuring out how to do more with fewer parts, really dialing in my understanding of low power design, and building a community firmware that folks are already contributing to. These are all lessons that are going to make the Open Book better when I return to it. Alas, every path one chooses to walk involves choosing not to walk down another, at least for a time.

micnolmad, 2022-07-13:
Thank you for the info. Just found this project and was so hyped and then not when it was "dead".. Glad to hear it is not.
Are you doing everything your self or do you have help? Because I am as many others also a tinkerer and have many projects that are very not complet simply down to the fact that these kinds of workloads are usually too much for one person.. at least it is for me having a family also.

TLDR; Get people to help you :)

commit f22c9e6d (2022-12-07): Project reboot; closes #1, closes #8, closes #9, closes #10, closes #11, closes #15, closes #18, closes #20, closes #22, closes #23, closes #24, closes #25, closes #26, closes #27, closes #28, closes #31, closes #33, closes #35, closes #39, closes #40, closes #42, closes #44, closes #47, closes #48, closes #49, closes #50, closes #52, closes #56, closes #58, closes #59, closes #65.
closed 2022-12-07 by joeycastillo