joeycastillo/The-Open-Book #60: About The battery

luka177, 2022-10-12:
I am willing to add li-ion/li-po BMS to this board, but have few concerns, firstly modern lithium batteries are no joke, so it should be done the right way. We need to add at least 2 ics, one for charging (possibly MCP73831 or BQ24040) and one to protect battery from over-discharge (MIC2776?). But that would add a lot components to solder, we may do universol board so one have to solder either aaa battery bay or lithium stuff. What do you think?

thargas, 2022-10-12:
Personally, I'd prefer to lose the AA thickness. I'm unsure how a universal
board avoids having to solder the lithium components, but I'm in favor of
minimizing the soldering as this would be my first of any kind. I'm
thinking maybe I should pick up the iron fist and get some practice.

luka177, 2022-10-12:
No no you got me wrong. You SHOULD NEVER EVER solder liion. Just dont. battery will be connected with some king of connector and glueed with double sided tape. But you will need more soldering to actually solder components needed to charge that battery properly (so no fire and explosions :D)

thargas, 2022-10-12:
Thanks for the elaboration and warning. I was meaning soldering the components needed for adding lithium batteries, not the batteries, but best to make sure I'm not about to do something stupid. :-)

joeycastillo, 2022-10-12:
You should feel free to do it! But a word of warning, I’m unlikely to want to merge in such a design. I have an idea of several other changes I’d like to make to realize a lipo powered version, and it will likely end up being a fairly different gadget. 

It’s open source hardware so you can absolutely do this in a fork and I think it would be rad! But when it comes to the designs in this repository, I do have a particular vision in mind (even if it’s something I may not be able to get to until the first half of 2023).

closed 2022-10-12 by joeycastillo