joeycastillo/The-Open-Book #68: Any place you can buy a presoldered version yet?

Kuratius, 2023-03-03:
Would be quite interested in this, particularly with a pico w. Doesn't need to be completely assembled.

DonnerWolfBach, 2023-03-03:
Same here - though I am outside of the us.

Kuratius, 2023-03-23:
I think the closest thingay be a kobo nia with a custom OS, but it's not easy to repair, though the storage can be replaced

Kuratius, 2023-04-06:
Seems to be similar, albeit with some differences.
A color panel might also be worth considering for a future iteration.
https://shop.pimoroni.com/products/inky-frame-5-7?variant=40057850331219

Kuratius, 2023-04-20:
May also be similar: https://pine64.com/product/pinenote-developer-edition/

Kuratius, 2023-04-27:
https://shop.oddlyspecificobjects.com/products/open-book-abridged-edition-kit

apparently this exists?

jjshoe, 2023-05-05:
Would love to be notified when a pre-soldered version is available

Kuratius, 2023-06-14:
may also be similar
https://github.com/atomic14/diy-esp32-epub-reader

chloe-everhart, 2023-09-16:
I'm also interested in a pre-soldered version.

Kuratius, 2023-10-12:
https://thepihut.com/products/4-2-e-paper-display-module-for-raspberry-pi-pico-black-white-400x300

Kuratius, 2023-10-12:
https://thepihut.com/products/3-7-e-paper-e-ink-display-for-raspberry-pi-pico-480x280

Just some stuff I found which is almost there

chloe-everhart, 2023-10-13:
I'm a person who would rather pay a maker for a completed device than pay for centralized assembly, and probably will not make a device myself, but will, if possible, use it to hand type old books that are too irregular for OCR to function well.