joeycastillo/The-Open-Book #70: Backlit e-ink

Taxicletter, 2023-04-02:
Can you make this with a backlit screen? 
I can imagine you would need extra parts like an on/off button for that.

Kuratius, 2023-04-02:
> Can you make this with a backlit screen? I can imagine you would need extra parts like an on/off button for that.

Eink screens aren't backlit because it's not possible to do so in a sensible manner. You can't do it without making the screen unreadable.
Even professional devices (kindle, Kobo) have a front lit screen because it makes more sense than a backlight.

Taxicletter, 2023-04-02:
Thank you. I misfrased my question, I now realise.
Lit in anyway I meant. So is frontlit possibly?

Kuratius, 2023-04-02:
> Thank you. I misfrased my question, I now realise. Lit in anyway I meant. So is frontlit possible?

If you don't care about even lighting then it should be easy, if if you do care about it then it's likely complicated. 

See this for some background info
https://goodereader.com/blog/electronic-readers/the-evolution-of-front-lit-e-reader-displays

Taxicletter, 2023-04-03:
Thank you very much for this clear and comprehensive answer!

Kuratius, 2023-04-29:
> Thank you very much for this clear and comprehensive answer!

I just learned that the manufacturer that makes the eink display *claims* to make touchscreen layers and front light layers compatible with the display, but I haven't seen a place where it's actually possible to buy them.

Taxicletter, 2023-05-03:
I bought a "regular" reader, when that breaks, hopefully in a long while, maybe lighting is solved with the Open Book and I can make one myself! :-)