joeycastillo/The-Open-Book #72: Boards dimensions to order at JLCPCB?

rodrirochad, 2023-06-12:
First time ordering a PCB service here. Sorry for the dumb question.

I have uploaded the .zip file as indicated in the "State of the Book" section, set to 1 mm and two-layer PCB, but I have no idea what to put in the board dimension field.
Where can I find the board dimension?

Thank you!

besselfunct, 2023-06-12:
By my measurements, the board is 85mm x 115mm, but you should check with your specific Gerber files.

If you have KiCAD, then you can open it up, and then use the "Gerber Viewer" app, and then open the "Edge Cuts" file. Set the scale to something like 1.0mm/div or 0.5mm/div, and then use the measure tool to get the board dimensions. There may be an easier way, but that's what I used.

rodrirochad, 2023-06-13:
Thank you for the help!
I have tried to upload the gerber files again and now it is catching the board dimensions automatically. I don't know what was happening before.

Though, I confirm that the main board is 85 mm x 115 mm and the Castellated E-Paper Driver is about 17 mm x 24 mm

closed 2023-06-13 by rodrirochad