joeycastillo/The-Open-Book #75: Fail flashing babel.bin

rodrirochad, 2023-07-03:
I have just built The-Open-Book and I am really excited that right in the first try I see things appearing on the screen. However, it seems that I have a problem flashing the language data (babel.bin).

First of all, because the flash chip "GD25Q16C" was out of stock, I am using is the "[Zetta ZD25Q16CTIGT](https://www.lcsc.com/product-detail/_Zetta-_C5160658.html)" instead. 
I thought it was compatible, but after everything I have tested, I am starting to think this is the cause.
I am using a 16Gb SD Card and the firmawares were downloaded from the links I found in the [Documentation](https://www.oddlyspecificobjects.com/projects/openbook/)

Please see description of the issue and things I have tried below:

I am facing the following problem:
1- At boot the screen shows the typical message: "Flashing the language chip", but NO progress bar is shown
2- After that, the screen shows only the library icon and battery voltage + battery icon. NO book list is shown (see picture)
[image]
3- Everytime I reboot the same thing happens.
4- The buttons are responding. The lock button, for instance, locks the book, but the “Slide the power switch to continue” message does NOT appear.
 
Things I have tried so far whithout success:
1- Removing the babel.bin from the sd card. The book complains that the babel.bin is missing. That is why I think the flashing failed
2- Replacing the flash chip for a new one.
3- Use a new SD card
4- Check electrical continuity or short circuits

I would be glad if you could provide insights on how to proceed or what should I try next.
Let me know if I missed any important information.

Cheers!

joeycastillo, 2023-07-08:
It sounds like this Flash chip may be incompatible. 

The lack of progress in the progress bar signals to me that the data never managed to get copied to the Flash chip, although it is interesting that it subsequently makes it to the list of books. Still, once there, the lack of text on the screen signals that it's not getting any data from the Flash chip; all the font data is stored on the chip, so text would fail to show up in this case. The battery voltage is displayed using a smaller character set that's embedded in the Libros application, hence why it shows up even if the other text does not.

rodrirochad, 2023-07-08:
Thanks!
I think so. After testing several SD cards and replacing the chip twice, I still have the same issue.
I have even tried to build a second book only to find the same issue.

Now I am waiting some GD25Q16C flash chip I have ordered from aliexpress to arrive.
I will keep you updated.

**Update on things I have tried and found investigating this issue:**

- Something interesting that adds up to the generic chip beeing the culprit, is that after desoldering the flash chip from the boad the book initializes in the same way.

- I have recently flashed the libros firmware forked from https://github.com/nvts8a/libros
And the openbook.log file output below points out to the problem starting Babel.

 /./_ __   _
///_///_/_\  v0.9.4

[INFO]  Started the SD Card service
[INFO]  Starting Babel
**[ERROR] Failed to start Babel.**
[WARN]  Library version file missing. Creating a new one at: /_LIBRARY/_VERSION
[INFO]  Looking for files in / to copy to /BOOKS...
[WARN]  alex-sese-dear-captain.txt found in the root of the SD card. Moving it to: /BOOKS/alex-sese-dear-captain.txt
[INFO]  Completed looking for and moving files that belong in /BOOKS
[INFO]  Processing new text files...
[INFO]  Processing new BookRecord /BOOKS/alex-sese-dear-captain.txt: ae0d9abd3b5b6ee09237c68a44afd1010b132b791724e92ea2047e2d21edbb5
[INFO]  Completed processing new BookRecord /BOOKS/alex-sese-dear-captain.txt: ae0d9abd3b5b6ee09237c68a44afd1010b132b791724e92ea2047e2d21edbb5
[INFO]  Completed processing new text files.
[INFO]  Writing new BookRecords to disk...
[INFO]  Completed writing new BookRecord files to disk.
[INFO]  Starting pagination of /BOOKS/alex-sese-dear-captain.txt...
[INFO]  Pagination of /BOOKS/alex-sese-dear-captain.txt completed.

rodrirochad, 2023-07-21:
I have managed to flash the library using the GD25Q16CSIG chip that I found in Aliexpress.
This chip has a larger footprint (SOP-8-208mil) but after carefully bending its terminal pins, it will fit in the PCB.

Closing the issue.

closed 2023-07-21 by rodrirochad