joeycastillo/The-Open-Book #76: Can we upload the construction video outside of Twitter?

alexjyong, 2023-07-25:
The video here https://twitter.com/i/broadcasts/1OyKAVPjrvaGb 

I don't mind going to Twitter, but with Twitter going down in flames, and with a good number of people not wanting to give Elon clicks, might be useful to have this posted elsewhere?

toddhull, 2023-07-28:
> For anyone else viewing this, I have put the video here in the meantime:
> https://sendvid.com/3g9i3969

It it possible to get the link to the video without porn advertised at the
bottom of the screen?

alexjyong, 2023-07-31:
@toddhull  I'm using ublock origin, so I didn't even see those ads. 😅 

I could upload this to Youtube. Just seemed self-serving to put it on my own account.

alexjyong, 2023-08-19:
Update. Having issues verifying my account and I've been busy with other crap. I'll try to get to this though.

Hermidion, 2023-09-27:
ayyo just upload it to Peertube