joeycastillo/The-Open-Book #80: GDEW042T2 module sold out everywhere

DeflateAwning, 2023-10-25:
Any advice on compatible modules? Or any plans to create a new rev that supports another display?

lastlich, 2023-10-27:
I did a very quick and dirty test and you could probably try using E2417CS0D1 or E2417PS0D1 if that is easier to find in your region. They seem to work, but they do have the flex cable coming out in a different direction so you would need to account for that.

berkecankocyigit, 2023-11-02:
Yes, I am having the same problem here. It is not available anywhere. Good Display doesn't have anything on their website either.

DeflateAwning, 2023-11-13:
Are there any compatible models from this store: https://goodisplay.aliexpress.com/store/600281

Would this model be a drop-in replacement? https://www.aliexpress.com/item/1005004825249437.html

smkj312, 2023-12-17:
THE WFH0420CZ35 CAN ALSO DRIVE.

d1nuc0m, 2024-04-04:
See also #77 and [branch newscreen](https://github.com/joeycastillo/The-Open-Book/tree/newscreen).

> Would this model be a drop-in replacement?

No, it uses a different IC (SSD1683  vs UC8176)