joeycastillo/The-Open-Book #85: [FEATURE REQUEST] audio player

SnowCat007, 2024-01-06:
I've always wished for an ereader that had a built in audio player since the only things I use a mobile device for nowadays are reading ebooks and manga, and listening to podcasts and music. If there was a way to add this to this project it would really be cool!
I don't really know what all needs to be added to make it work, but I know that a 3.5mm headphone jack would make it even better; though I also know some people would probably want bluetooth nowadays if audio was added. The other reason this would be cool would be if audio books were ever made as easy to move from device to device as ebooks are with the epub and pdf formats.
Anyways, thanks for the great work you've put into this project already, it's turing into a really awesome ereader option for people who want one that isn't connected to any ebook service like kindles are, and I'm looking forward to seeing where it goes in the future. :)

nerdguy1338, 2024-01-06:
Regarding audio book transfer, if you would like to free your Audible library to use it on any mp3 player, this tool is fantastic.
https://github.com/rmcrackan/Libation

Not made by me, I just love it.

SnowCat007, 2024-01-06:
Thank you for sharing this with me, I'll definately be using this with any audio books from kindle. Hopefully we'll see a day when apps like this aren't needed as much.