joeycastillo/The-Open-Book #86: e paper display sold out

the-kant-rawls-mafia, 2024-02-05:
the epaper display recommended on the website that goes with the PCB is sold out. Would this display work:

https://www.good-display.com/product/386.html

???

Silent-Hunter, 2025-02-20:
What happens if you don't have partial refresh or greyscale mode? If the book only works with plaintext files, wouldn't those things not be needed? Please pardon my ignorance.

Kuratius, 2025-02-20:
Partial refresh and greyscale can be added retroactively if you can edit the waveform data inside the display. Even black and white displays have at least 4 waveforms available that you can use.
Otherwise yes you'd have to live with black/white and dithering.

joeycastillo, 2025-02-20:
Partial refresh is pretty crucial to having a responsive UI. Like when selecting a book, waiting a full second between pressing a button and moving the focus one row is not delightful. Whereas with partial refresh it’s a lot quicker. It’s definitely solvable, just no longer an area of focus for me since this is a design I never shipped or recommended that folks try and build.

Silent-Hunter, 2025-02-21:
Well unfortunately the display you designed for is unavailable anywhere, I think it's been discontinued. So not having partial refresh is definitely preferable to not having an Open Book at all. Is there another display you had in mind?

Kuratius, 2025-02-21:
For what it's worth, except for the size a pimoroni badger 2040 has essentially the same functionality as what you would get from the open book, just a smaller screen size. You need a special driver to get greyscale, though. If you were looking to write software that can later be used with the open book, this would be a good candidate. It also avoids having to do your own assembly.
 https://shop.pimoroni.com/products/badger-2040-w?variant=40514062188627

Silent-Hunter, 2025-02-24:
Definitely some interesting ideas. I really want something that can use standard AA or AAA batteries, that way I can just swap them when they get low, and easily replace them when they stop reacharging. That's one of the things that really drew me to this project.

Kuratius, 2025-02-24:
Badger2040 uses AA batteries I think.

Silent-Hunter, 2025-02-24:
Yeah I looked at it, but the screen is just too low res for a book I think.